pimoroni/hyperpixel4 #201: HQ camera with hyperpixel 4.0 square (using the overlay) not detected

feiticeir0, 2023-05-09:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

What's gone wrong?
libcamera-hello doesn't detect the HQ camera. 
Here's the output from libcamera-hello:
```
pi@raspberrypi:~ $ libcamera-hello
Made DRM preview window
[0:00:39.044057716] [751]  INFO Camera
```

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
Using the overlay added to /boot/config.txt
```
[all]
dtoverlay=vc4-kms-dpi-hyperpixel4sq

```
**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
I have the Square version
3. Touch or non Touch?
It's the touch version
5. Approximately when was it purchased
March 18th, 2022
7. How is it connected to your Pi
GPIO

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
```
pi@raspberrypi:~ $ cat /proc/cpuinfo  | grep -i revision
CPU revision    : 3
CPU revision    : 3
CPU revision    : 3
CPU revision    : 3
Revision        : b03111
```
3. Which OS release are you using: `lsb_release --description` and `uname -r`
```
pi@raspberrypi:~ $ lsb_release --description
Description:    Raspbian GNU/Linux 11 (bullseye)
```
```
pi@raspberrypi:~ $ uname -r
6.1.21-v8+
```

5. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Fresh installation from about 1h ago..

**Extra debugging information**

I've already tried the solution[ from this closed issue](https://github.com/pimoroni/hyperpixel4/issues/188), but no luck.

Thank you

feiticeir0, 2023-05-11:
So, no one from Pimononi has a clue what's happening ? ...  I can't use the HyperPixel with the HQ camera ? Or any other camera ?

kisdia, 2023-05-12:
What is the printout of
`dmesg | grep "i2c"`

feiticeir0, 2023-05-13:
> 

Hi ! 
Thank you for the reply.

Here's the result from the command:
```
[    0.079820] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/imx708@1a
[    5.099113] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    5.099276] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    5.140317] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)
[    5.732214] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    5.744728] i2c i2c-23: Added multiplexed i2c bus 0
[    5.746819] i2c i2c-23: Added multiplexed i2c bus 10
[    5.757530] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[    7.615262] input: EP0110M09 as /devices/platform/i2c@0/i2c-22/22-0048/input/input6
[    8.340177] dw9807 10-000c: I2C write CTL fail ret = -121
[    8.353944] dw9807 10-000c: I2C write STATUS address fail ret = -121
[    8.353974] dw9807 10-000c: dw9807_ramp I2C failure: -121
[    8.355392] dw9807 10-000c: I2C write STATUS address fail ret = -121
[    8.355423] dw9807 10-000c: dw9807_ramp I2C failure: -121
[    8.355774] dw9807 10-000c: I2C write CTL fail ret = -121
```
I'm using a Raspberry PI 4

feiticeir0, 2023-05-13:
I've solved by adding the following lines to /boot/config.txt

`
dtoverlay=vc4-kms-dpi-hyperpixel4sq
disable_poe_fan=1
force_eeprom_read=0
camera_auto_detect=0
dtoverlay=imx477
`

I'm now able to use the camera and the Pimoroni HyperPixel

closed 2023-05-13 by feiticeir0