pimoroni/hyperpixel4 #205: Hyperpixel 4.0 Square (No Touch) Screen Scrolling / Ghosting

0xMisterBailey, 2023-10-09:
Upon Installing the legacy drivers (option '1' for 'weirdly square' 'no-touch' and rebooting. Display shows boot splash screen just fine, but at the desktop the image "scrolls" like a only CRT monitor, and the display is "ghosted" showing artifacts. 
--
Product Purchased: **HyperPixel 4.0 Square - Hi-Res Display for Raspberry Pi - Non-Touch - PIM475**
Purchase Date: Sept 2023
Connection Type: GPIO with extender

My Pi: Raspberry Pi 4-B
OS: Debain Version: 11 (Bullseye), kernal version 6.1
--

[See here for YouTube video showing the issue. ](https://youtube.com/shorts/KtA3p7mYmzk?si=9ZIbNOlnmFQod8bB)

Anyone have any solutions for this? I am assuming this is not a hardware issue, as the splash screen displays just fine.

0xMisterBailey, 2023-10-09:
Currently installing a fresh image of Buster and starting from scratch...

0xMisterBailey, 2023-10-10:
Fresh image of BUSTER and reinstall of the hyperpixel '-legacy' curl using options [1} (Pi 4 Weirdly sqaure] and [0] (manufactured after 2021) solved the issue.

TiborUdvari, 2024-05-17:
I have the same issue with bookworm and raspberry pi 4, trying your solution

closed 2026-08-25 by Gadgetoid