pimoroni/hyperpixel4 #206: Screen rotation in X and Bookworm

addohm, 2023-10-15:
I'm firing up two new servers and the current method of install instruction just doesn't work for me.

In the terminal it absolutely works as expected. 

In X I can't seem to get it to rotate using any of the suggestions in the issues or googling.  

Honestly, I've never had a smooth experience with starting up a new pi with a hyperpixel screen.  The basic premise of simply adding dtoverlay and dtparam just never works for any window manager.  Is it possible to add instructions to get it to rotate (including touch) in WM's as well?  

Previously I ran all my hyperpixels on Pi3A's and just blanket copied a boilerplate image I put together.  Now that I am trying to update everything (hardware and software), it's really testing my patience >.<

addohm, 2023-11-09:
Is there _any_ dev happening currently for bookworm or is this a dead project?

Gadgetoid, 2023-11-09:
HyperPixel 4 rectangular is supported in Pi OS and can be rotated with the display settings in (at least on Pi 4/5 afaik).

Outside of this, and short of playing telephone and raising issues with Raspberry Pi, I can't do any meaningful development anymore. Rotation is something that requires a GPU or software alternative at some point to work, and every bit of GUI software, window manager or otherwise has its own particular nuisances (ha, sic) that are not only not consistent between software, but not consistent between Pi versions, OS versions or anything else.

*That said* you can try my - probably now broken again - rotation utility from here: https://github.com/pimoroni/hyperpixel4-rotate if it doesn't work, poking at the internals (it just sets config files) might give you some clues.

And yeah, there's dev happening for bookworm- on the 150 other things that have been broken and I need to fix 😭  HyperPixel 4 (rectangular, at least) is low on that list because - from my brief tests on a vanilla Pi OS - it's actually in a better place than it's been for years, and supporting anything else is apparently a fool's errand.

addohm, 2023-11-09:
> That said you can try my - probably now broken again - rotation utility from here: https://github.com/pimoroni/hyperpixel4-rotate if it doesn't work, poking at the internals (it just sets config files) might give you some clues.

Are you saying that I should try the legacy version?  I've been avoiding it because even on bullseye it was almost always trouble for me personally.  

> HyperPixel 4 (rectangular, at least) is low on that list because - from my brief tests on a vanilla Pi OS - it's actually in a better place than it's been for years, and supporting anything else is apparently a fool's errand.

My initial tests of vanilla pi os (bookworm, 64 bit) is what brought me here.  It doesn't work at all out of the box and even the window manager doesn't show it.  It does back light (I'm assuming its on its own power) but nothing gets displayed either in the text or desktop environment.

It might sound like I'm pooping on the product but I'm not.  I love this and want to continue to see it thrive and if I didn't feel like I was constantly wasting my time with raspberry pis I'd certainly make contributions.

Gadgetoid, 2023-11-09:
> It might sound like I'm pooping on the product but I'm not. I love this and want to continue to see it thrive and if I didn't feel like I was constantly wasting my time with raspberry pis I'd certainly make contributions.

NOO KILL IT WITH FIREEEEE!!!!!! :sob: :sob:

What Pi are you running Bookworm/64bit on? On a Pi 4/400/5 just adding `dtoverlay=vc4-kms-dpi-hyperpixel4` to `/boot/config.txt` should be all you need. I tested this when Bookworm was in beta, and fed back some tweaks to get our touchscreen recognised by the Screen Configuration utility. Worked like magic. I guess it could have broken since!

On Pi 3... I'm, uh, haven't tried.

addohm, 2023-11-09:
> > It might sound like I'm pooping on the product but I'm not. I love this and want to continue to see it thrive and if I didn't feel like I was constantly wasting my time with raspberry pis I'd certainly make contributions.
> 
> NOO KILL IT WITH FIREEEEE!!!!!! 😭 😭
> 
> What Pi are you running Bookworm/64bit on? On a Pi 4/400/5 just adding `dtoverlay=vc4-kms-dpi-hyperpixel4` to `/boot/config.txt` should be all you need. I tested this when Bookworm was in beta, and fed back some tweaks to get our touchscreen recognised by the Screen Configuration utility. Worked like magic. I guess it could have broken since!
> 
> On Pi 3... I'm, uh, haven't tried.

Pi5 is the only one I'm referencing at this point.   I haven't set up a new Pi4 with Hyperpixel Touch 4.0 Rect in about a year.  But I do remember it took me like an entire day to get it going on a Pi4.  But...being 1y ago it's probably irrelevant anyway.

metanav, 2023-11-25:
I am in the same boat (Pi 5 / Bookworm (64bit)). 

/boot/config.txt:
```
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dpi-hyperpixel4
max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1
[cm4]
otg_mode=1
[all]

```

Linux raspberrypi 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

```
me@raspberrypi:~ $ dmesg | grep Goodix
[    7.230636] Goodix-TS 11-0014: supply AVDD28 not found, using dummy regulator
[    7.249882] Goodix-TS 11-0014: supply VDDIO not found, using dummy regulator
[    7.259695] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
[    7.287976] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
[    7.314298] Goodix-TS 11-0014: I2C communication failure: -6
[    7.314606] Goodix-TS 11-005d: supply AVDD28 not found, using dummy regulator
[    7.314720] Goodix-TS 11-005d: supply VDDIO not found, using dummy regulator
[    7.317903] Goodix-TS 11-005d: ID 911, version: 1060
[    7.399375] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input3
```

```
me@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

mislav, 2023-12-02:
@addohm I've shared instructions that worked for me here: https://github.com/pimoroni/hyperpixel4/issues/177#issuecomment-1837186505

Note that since around the time that you've started this thread, the official Raspberry Pi Debian image does not boot the X11 window system, but Wayland/Wayfire instead. For me, that is why editing any X11 settings didn't have effect in the desktop environment. Editing `wayfire.ini` is what eventually solved it.

closed 2026-08-25 by Gadgetoid