pimoroni/hyperpixel4 #208: Hyperpixel 4.0 Touch RPi 5 [Bookworm (64bit)] not working

metanav, 2023-11-25:
The Hyperpixel 4.0 Touch display is blank after booting the RPi 5 / Bookworm (64bit). 

/boot/config.txt:
```
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
**dtoverlay=vc4-kms-dpi-hyperpixel4**
max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1
[cm4]
otg_mode=1
[all]

```

Linux raspberrypi 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

```
me@raspberrypi:~ $ dmesg | grep Goodix
[    7.230636] Goodix-TS 11-0014: supply AVDD28 not found, using dummy regulator
[    7.249882] Goodix-TS 11-0014: supply VDDIO not found, using dummy regulator
[    7.259695] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
[    7.287976] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
[    7.314298] Goodix-TS 11-0014: I2C communication failure: -6
[    7.314606] Goodix-TS 11-005d: supply AVDD28 not found, using dummy regulator
[    7.314720] Goodix-TS 11-005d: supply VDDIO not found, using dummy regulator
[    7.317903] Goodix-TS 11-005d: ID 911, version: 1060
[    7.399375] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input3
```

```
me@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

shbisson, 2023-11-29:
Same here with a Square.

metanav, 2023-11-30:
By the way, it works with Ubuntu 23.10 so I can confirm the display is fully functional. But, of course, I would like to use it with the RPi OS Bookworm (64bit).

metanav, 2023-12-03:
It works!  https://github.com/pimoroni/hyperpixel4/issues/177#issuecomment-1837186505

closed 2023-12-03 by metanav