pimoroni/hyperpixel4 #212: Hyperpixel 4 no display on Pi 400

nofurthur, 2024-02-24:
**Describe the bug**

Hyperpixel 4 not displaying image on Raspberry Pi 400 - either displays backlight only or displays white screen and then backlight only.

**To Reproduce**

I have attempted installs from the one line installer and from Github on fresh installs of both the default Raspberry Pi OS, and the Bullseye distro identifed in the Hyperpixel4 Rectangular readme. 
**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or **Rectangular**?
2. Touch or **non Touch**?
3. Approximately when was it purchased - 2/20/24
4. How is it connected to your Pi - have attempted both directly and with adafruit Cyberdeck

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision        : c03130
2. Which OS release are you using: `lsb_release --description` and `uname -r`
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Completely fresh each time

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

[    9.265800] Goodix-TS 22-0014: supply AVDD28 not found, using dummy regulator
[    9.266739] Goodix-TS 22-0014: supply VDDIO not found, using dummy regulator
[    9.344535] Goodix-TS 22-0014: Error reading 1 bytes from 0x8140: -6
[    9.370561] Goodix-TS 22-0014: Error reading 1 bytes from 0x8140: -6
[    9.397581] Goodix-TS 22-0014: I2C communication failure: -6
[    9.398315] Goodix-TS 22-005d: supply AVDD28 not found, using dummy regulator
[    9.398561] Goodix-TS 22-005d: supply VDDIO not found, using dummy regulator
[    9.399508] Goodix-TS 22-005d: Error reading 1 bytes from 0x8140: -6
[    9.426726] Goodix-TS 22-005d: Error reading 1 bytes from 0x8140: -6
[    9.453585] Goodix-TS 22-005d: I2C communication failure: -6

And check i2c is working:

`ls /dev/i2c-*`

/dev/i2c-20  /dev/i2c-21  /dev/i2c-22

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Additional info - I have attempted commenting out dtoverlay=vc4-kms-v3d in /boot/firmware/config.txt and though this does present a white screen at boot, I see no image. 

Additionally I am unable to execute rotate commands - have attempted to add rotate info to /boot/firmware/config.txt and this does not help the issue.

Daenks, 2024-03-28:
Hi, having this exact same issue. Did you ever find a solution, @nofurthur ?

nofurthur, 2024-03-28:
Unfortunately I could not resolve the issue.

On a few occasions I was able to get the display to produce an image by
using a fresh copy of the specific build that Pimoroni recommends from
RaspPi Imager.

However this image was always inverted on both X and Y axis, and not
properly aligned/calibrated with the display itself cutting off most of the
image. I spent several hours attempting different recommended commands and
configurations to correct the image inversion and misalignment. At no point
was I able to correct any aspect of the displayed image, not even
temporarily.

As the display (purchase brand new from an OEM partner) is pretty much
unusable in its current state for me (either only displaying backlight or a
totally misaligned display) I eventually gave up on the Hyperpixel and
resigned myself to using the HDMI port and a standard monitor... which is
working perfectly.

Sorry that I can't be more help as I know this issue was extremely
frustrating for me considering how well the Hyperpixel seems to be reviewed
and the fact that it's on the pricier side.

Daenks, 2024-03-28:
Thanks for taking the time to reply. I will keep trying for a while.

nofurthur, 2024-03-29:
Sorry I couldn't be of more help. Please update me if you find any
solutions.

closed 2026-08-25 by Gadgetoid