pimoroni/hyperpixel4 #215: Getting touch rotation to work on Raspberry OS Bookworm/Wayland

sentient-cloud, 2024-05-25:
Not sure where to put this, since I can't find a matching issue where this was still unsolved.

My setup is a RPi 3 B+ with a rectangular 480x720 touchscreen in inverted mode, the display rotation can be changed and persisted through the screen preferences menu, but this did not rotate the touch input as well.

I found a solution, which involves adding a udev rule.

Create `/etc/udev/rules.d/98-hyperpixel4-calibration.rules` and add ONE of the following rules to it, depending on your configuration:

```
# left
ATTRS{name}=="11-005d Goodix Capacitive TouchScreen", ENV{LIBINPUT_CALIBRATION_MATRIX}="0 -1 1 1 0 0 0 0 1"

# right
ATTRS{name}=="11-005d Goodix Capacitive TouchScreen", ENV{LIBINPUT_CALIBRATION_MATRIX}="0 1 0 -1 0 1 0 0 1"

# normal
ATTRS{name}=="11-005d Goodix Capacitive TouchScreen", ENV{LIBINPUT_CALIBRATION_MATRIX}="1 0 0 0 1 0 0 0 1"

# inverted
ATTRS{name}=="11-005d Goodix Capacitive TouchScreen", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 -1 1 0 0 1"
```

Then reload the udev rules (applies the rule immediately, instead of having to reboot):

```bash
sudo -Es
udevadm control --reload-rules && udevadm trigger
exit
```

Gadgetoid, 2024-05-30:
You should be able to right click on the screen in preferences and assign a touchscreen so that rotation is updated automatically.

This… isn’t very obvious and we need to document it somewhere 🫠

sentient-cloud, 2024-06-04:
I did try that, but didn't work.

Gadgetoid, 2024-06-06:
Seems to reliably fail for me first time and then decide to work after a reboot. I’m not sure why. I need to dig in to arandr to see what config it’s actually changing and maybe produce an updated version of https://github.com/pimoroni/hyperpixel4-rotate

aiui Pi are abandoning wayfire for something else (Lawbc) so the goalposts are going to move *again* 🫠

closed 2026-08-25 by Gadgetoid