pimoroni/hyperpixel4 #216: HyperPixel 4.0 on RPi4 - LCD working but touch inverted

jdschuitemaker, 2024-05-30:
Yesterday wanted to use this hardware for the first time since I have had it for a long time laying in a drawer. Unaware of possible difficulties I flashed a new image of the latest OS. After running some scripts and changing settings in  /boot/firmware/config.txt the screen was showing it's image.

There is a problem with the touch though. When I use my fingers to move the cursor it is drawing a white area, similar to the behavior of pressing the left mouse button while moving the cursor. The mouse that I attached works as expected, but how can I get touch working similarly?

Another weird thing is that when I set `dtoverlay` to a HyperPixel specific setting the screen doesn't show an image. Just blacklight. So I currently have it at the default setting:

`
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
#dtoverlay=vc4-kms-dpi-hyperpixel4
#dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x
#dtparam=rotate=90, touchscreen-inverted-x
max_framebuffers=2
`

Gadgetoid, 2024-05-30:
The Raspberry Pi desktop “Screen Configuration” utility has a drop down menu that lets you assign a touchscreen to a specific screen- that should, in theory, take care of rotation.

Gadgetoid, 2024-05-30:
“lets you assign a _touchscreen_ to a specific screen”?

Gadgetoid, 2024-05-30:
It’s buried in this menu (not shown here because I don’t have a touchscreen attached), and IIRC takes care of mapping touch input to the related display-

[image]

jdschuitemaker, 2024-05-30:
Yes, I have that menu. Are you referring specifically to the Orientation setting? That shows "22-0014 Goodix Capacative TouchScreen" with a checkmark. Also the Active option that you show has a Checkmark, just like your screenshot.

jdschuitemaker, 2024-05-30:
https://github.com/pimoroni/hyperpixel4/assets/2353413/00fc509a-4fb7-4001-a958-21c5307efa87

This video demonstrates the issue. First I show how it works with the mouse, then I show how I touch the screen and the moue cursor disappears. Touch, double touch do not reveal the mouse. Touch and move draws a rectangle.

Gadgetoid, 2024-05-30:
Do you have any additional *touch* rotation/translation settings in device tree, or have made any efforts to configure touch rotation/translation elsewhere?

The last time I checked this (granted the goal posts seem to move very regularly) just assigning the touchscreen to the display in Screen Configuration was enough to get it synced up and correctly mapped to the display.

Also, try rotating the screen and rotating it back again. Screen Configuration is... uh a very bolted-on solution that mostly drops config files in the right places, but may not do a proper job of that unless it's poked.

jdschuitemaker, 2024-05-31:
Let me start with thanking you for your help. It is much appreciated!

The only thing I did was running the script mentioned in this post: [https://github.com/pimoroni/hyperpixel4/issues/177#issuecomment-1870357875](url)

I rotated the screen to all 4 positions, but touch behavior remains the same.

Gadgetoid, 2024-05-31:
Curses. I’ll try setting it up today and see what’s broken 🫠

Gadgetoid, 2024-05-31:
Okay I've just realized you're talking Pi 4 here and I'm poking on a Pi 5 so that probably isn't helpful. I'm pretty sure the instructions with respect to Screen Layout Editor should work the same on both, though it's been long enough since I tested all this that I might be kidding myself.

Everything seems to work great - albeit after a couple of reboots - on my Pi 5.

Currently booting up a Pi 4 to check my assertions before I make a further fool of myself.

jdschuitemaker, 2024-05-31:
No worries ;)

Just to be sure, I am on the 2024-03-15-raspios-bookworm-arm64 image with the RPi

Gadgetoid, 2024-05-31:
Okay it works basically the same on my Pi 4, as I recall, though it was also borked until I rebooted!

```
phil@phils-pi4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

```
phil@phils-pi4:~ $ uname -a
Linux phils-pi4 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
```

So assigning the touchscreen, rotating and rebooting *should* work. I suspect the install you ran has dropped some config file somewhere that's interfering with it.

Check `/udev/udev/rules.d/` for anything HyperPixel related.

Also your pasted config.txt shows all of the pertinent lines as being commented out. What do the relevant hyperpixel sections of `/boot/firmware/config.txt` look like?

jdschuitemaker, 2024-05-31:
I tried a new install but never got the HyperPixel to show an image. Do I need to have a HDMI screen connected for the first install?

Here is what I did:

- After first boot: apt get update and upgrade
- Added the two lines to /boot/firmware/config.txt:
```
 dtoverlay=vc4-kms-dpi-hyperpixel4 
 dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x
```

After reboot no image. Then I ran the script from [this](https://github.com/pimoroni/hyperpixel4/issues/177#issuecomment-1870357875) post and rebooted, still no image.

Gadgetoid, 2024-05-31:
Try removing the "dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x"

You should not need the dtparam anymore, since rotation and touch calibration is handled by the OS.

Make sure I2C and any other interfaces are disabled, though I don't think these are enabled by default.

The *only* config change you should need to make is adding `dtoverlay=vc4-kms-dpi-hyperpixel4` to `/boot/firmware/config.txt`.

jdschuitemaker, 2024-06-02:
Ok, another try and no succes. I did a new install with an HDMI monitor connected. Then verified if the interfaces (I2c, etc) were disabled, which was true. Just SSH was enabled.

Then, while the HDMI monitor is still connected, modified the setting for `dtoverlay` in `/boot/firmware/config.txt` file to this

```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-dpi-hyperpixel4
max_framebuffers=2
```

Then I did a shutdown (power off) of the RPi4, disconnected HDMI and rebooted a couple of time by connecting over SSH. Still no image on the HyperPixel4 screen.

Gadgetoid, 2024-06-03:
Ooof, I don't know how we've arrived at a *worse* outcome. Is the board seated directly and securely on top of the Pi?

daveic, 2024-06-08:
> Ok, another try and no succes. I did a new install with an HDMI monitor connected. Then verified if the interfaces (I2c, etc) were disabled, which was true. Just SSH was enabled.
>
> Then, while the [...]

Hi,
in the '/boot/firmware/config.txt` file you should have both these lines:
dtoverlay=vc4-kms-dpi-hyperpixel4
dtoverlay=vc4-kms-v3d

do not remove the already present dtoverlay=vc4-kms-v3d, just add the new one.

jdschuitemaker, 2024-06-19:
Sorry, forgot all about this. The screen is working, with TV connected to HDMI and the HyperPixel, but also with nothing connected to HDMI and just the HyperPixel.

The whole problem for me (from the start of this isse) has been that touch responses do not work as expected. There is no mouse pointer. What happens when I touch the screen with my finger and slide over the screen is that an area is selected while I expect a moving mouse pointer. I have to use an external mouse to be able to control. I am running a browser all day, showing an aircraft radar (from ADSB data) and it would be nice if it could be used without mouse.

closed 2026-08-25 by Gadgetoid