pimoroni/hyperpixel4 #217: Square - Instructions for Raspberry pi Zero W

Hangman228, 2024-06-16:
I had few problems with my HyperPixel 4 Square on my raspberry pi zero W so here is the instructions to get it works :

- You will need to solder a header to plug it in to the screen
- Install latest raspberry pi os 64 bits (2024-03-15) with raspberry pi imager
- Log to your raspberry with ssh and edit file : /boot/firmware/config.txt (sudo nano /boot/firmware/config.txt)
- Just add at the begining of the file "dtoverlay=vc4-kms-dpi-hyperpixel4" (you will find another dtoverlay line further in config file but leave it as it is)
- Now everything works :)

Yannick-hildenbrand, 2024-07-09:
So far I know, RPI Zero W is a single core 32-bit processor, that does not support the 64-bit version of RPI OS. Therefore, I wonder how you installed the 64-bit version, which is in theory not compatible (I'd like to live in theory, because in theory, everything works ^^). Additionnally, there are two hardware releases for the square display. The "old" one, before 2021 that, (according to Pimironi product page) does not work with the new drivers (after debian Buster), and the new one (marked "Hyperpixel XP" - see https://shop.pimoroni.com/products/hyperpixel-4-square?variant=30138251444307) that is compatible. From my POV, early hyperpixel HW variants cannot run under Bullseye or Bookworm versions. I finally installed the latest Buster version from the archive, 32-bit, combined with the legacy drivers (curl -sSL https://get.pimoroni.com/hyperpixel4-legacy | bash).

This is the only combination that worked for me.

If you have an old display (2019 in my case), follow the steps below:

- Download the latest version of RPI OS based on buster - I believe this is there (https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2021-05-28/)
- Install it on the Zero W, make sure SPI and I2C are disabled.
- Update the packages (sudo apt update, sudo apt upgrade ...)
- (optional) activate SSH, to get access with PuttY or KittY or any other SSH client you like
- Install the legacy drivers (curl -sSL https://get.pimoroni.com/hyperpixel4-legacy | bash)

This was the key for me.

This is not completely satisfying. I'd prefer an up to date configuration that works with the new drivers, but haven't found an answer to this, despite my searches on the web.

If someone has a better solution with the latest RPI OS versions, please let me know :)

Hope this helps !

Cheers.

Yannick.

Hangman228, 2024-07-13:
My bad, I wanted to write "Raspberry Pi Zero 2W"... didn't try on old raspberry pi zero W.... I will modify my post

Hangman228, 2025-01-21:
Now (21/01/25) this thing is outdated, just follow instructions from official page : [](https://shop.pimoroni.com/products/hyperpixel-4-square))
`If you're using a recent version of Raspberry Pi OS (Bullseye or later) then you'll need to use the built in kernel drivers - just add the following line to the end of your boot/firmware/config.txt (and then reboot):

dtoverlay=vc4-kms-dpi-hyperpixel4sq`

closed 2025-01-21 by Hangman228