pimoroni/hyperpixel4 #219: Does this work with RPi 5?

WadDyORedDy, 2024-07-10:
Followed the documentation from https://github.com/pimoroni/hyperpixel4/tree/pi4 and https://github.com/pimoroni/hyperpixel4/issues/177 but to no avail. 

I've edited the config.txt aswell, but no dice.

seandoyle, 2024-07-14:
Same question here about the RPi 5 - I see the splash screen when I reboot, it fades out, pops up again, then disappears. There is a small blinking cursor in the corner.

dmesg reports: (grepping for error lines only)
```
[    0.353593] bcm2708_fb: probe of soc:fb failed with error -2
[    2.879974] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
[    2.909672] Goodix-TS 11-0014: Error reading 1 bytes from 0x8140: -6
```
No errors in the boot log. 

Running Debian 12.6. 

Here's the release info:
```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Darwin MacBook-Pro-5.local 23.5.0 Darwin Kernel Version 23.5.0: Wed May  1 20:12:58 PDT 2024; root:xnu-10063.121.3~5/RELEASE_ARM64_T6000 arm64
```
uname -a 
```
Linux sidecar 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

seandoyle, 2024-07-21:
I've reinstalled the OS (`Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`) but the problem remains. 
I tried this installation 

`git clone https://github.com/pimoroni/hyperpixel4 -b pi4` 
but it fails even if I update the python gpio library using 
`sudo apt install python3-rpi-lgpio` 
``` 
journalctl -xeu hyperpixel4-init.service
░░ 
░░ A start job for unit hyperpixel4-init.service has begun execution.
░░ 
░░ The job identifier is 45145.
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]: Traceback (most recent call last):
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:   File "/usr/bin/hyperpixel4-init", line 241, in <module>
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:     setup_pins()
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:   File "/usr/bin/hyperpixel4-init", line 12, in setup_pins
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:     GPIO.setup([CLK, MOSI, CS], GPIO.OUT)
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:   File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 677, in setup
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:     initial = _check(lgpio.gpio_read(_chip, gpio))
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 903, in gpio_read
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:     return _u2i(_lgpio._gpio_read(handle&0xffff, gpio))
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]:     raise error(error_text(v))
Jul 20 19:49:06 raspberrypi hyperpixel4-init[88745]: lgpio.error: 'GPIO not allocated'
Jul 20 19:49:06 raspberrypi systemd[1]: hyperpixel4-init.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit hyperpixel4-init.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jul 20 19:49:06 raspberrypi systemd[1]: hyperpixel4-init.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit hyperpixel4-init.service has entered the 'failed' state with result 'exit-code'.
Jul 20 19:49:06 raspberrypi systemd[1]: Failed to start hyperpixel4-init.service - HyperPixel 4.0" LCD Display Initialization.
░░ Subject: A start job for unit hyperpixel4-init.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit hyperpixel4-init.service has finished with a failure.
░░ 
░░ The job identifier is 45145 and the job result is failed.
```

WeeJeWel, 2024-07-31:
I can confirm that at least Raspberry Pi OS Bookworm (Lite) works on a Raspberry Pi 5. I can see a terminal.

MacEdgar1, 2024-10-04:
The screen does work with the Raspberry PI 5.  Using min with Klipper.  Was having issues getting the touchscreen to be correct.

https://forums.pimoroni.com/t/hyperpixel4-screen-rotation-touchscreen-fix-how/14778/9

The Raspberry Pi config.txt is in /boot/firmware/...

This entry allowed me to rotate the screen so it's in landscape with the power cable on top.

dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y

WadDyORedDy, 2024-10-04:
Hi MacEdgar1 - thank you very much for your reply! 

Your dtoverlay-line combined with raspi-config > advanced > Wayland > select X11 made the screen show an actual image this time.

However, the "rotate" part did not seem to work, but with an image I could go to Preferences > Screen Configuration > Layout > Screens > DPI-1 > Orientation > Right (to get the power-cord on top).

Thank you very much for your comment! :-)

closed 2026-08-25 by Gadgetoid