pimoroni/hyperpixel4 #222: [QUERY] Compatibility With RADXA ZERO 3E

influential-eliot, 2024-09-10:
### Given The GPIO Is This SBC Potentially Compatible?
I'm wanting to work with the [Radxa 3E board](https://docs.radxa.com/en/zero/zero3?model=zero-3e) (_or it's 3W sibling_) as part of a '_wing mirror_' project for my old 928.

Are their new GPIO layouts and/or featuresets compatible with any of the Hyperpixel screens? (_hoping the roundy, but anything would be nice_)

#### GPIO
The below table is from [this page](https://docs.radxa.com/en/zero/zero3/hardware-design/hardware-interface#gpio-interface) on their Wiki.

GPIO number | Function5 | Function4 | Function3 | Function2 | Function1 | Pin# | Pin# | Function1 | Function2 | Function3 | Function4 | Function5 | GPIO number
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  |   |   |   |   | +3.3V | 1 | 2 | +5.0V |   |   |   |   |  
32 |   |   |   | UART3_RX_M0 | GPIO1_A0 | 3 | 4 | +5.0V |   |   |   |   |  
33 |   |   |   | UART3_TX_M0 | GPIO1_A1 | 5 | 6 | GND |   |   |   |   |  
116 |   | PWM14_M0 |   |   | GPIO3_C4 | 7 | 8 | GPIO0_D1 | UART2_TX_M0 |   |   |   | 25
  |   |   |   |   | GND | 9 | 10 | GPIO0_D0 | UART2_RX_M0 |   |   |   | 24
97 |   |   |   |   | GPIO3_A1 | 11 | 12 | GPIO3_A3 |   |   |   | I2S3_SCLK_M0 | 99
98 |   | I2S3_MCLK_M0 |   |   | GPIO3_A2 | 13 | 14 | GND |   |   |   |   |  
104 |   |   |   |   | GPIO3_B0 | 15 | 16 | GPIO3_B1 | UART4_RX_M1 | PWM8_M0 |   |   | 105
  |   |   |   |   | +3.3V | 17 | 18 | GPIO3_B2 | UART4_TX_M1 | PWM9_M0 |   |   | 106
147 |   | PWM15_IR_M1 | I2S3_SCLK_M1 | SPI3_MOSI_M1 | GPIO4_C3 | 19 | 20 | GND |   |   |   |   |  
149 | UART9_TX_M1 | PWM12_M1 | I2S3_SDO_M1 | SPI3_MISO_M1 | GPIO4_C5 | 21 | 22 | GPIO3_C1 |   |   |   | I2S1_SDO2_M2 | 113
146 |   | PWM14_M1 | I2S3_MCLK_M1 | SPI3_CLK_M1 | GPIO4_C2 | 23 | 24 | GPIO4_C6 | SPI3_CS0_M1 | PWM13_M1 | UART9_RX_M1 | I2S3_SDI_M1 | 150
  |   |   |   |   | GND | 25 | 26 | NC |   |   |   |   |  
138 |   | I2C4_SDA_M0 | I2S2_SDI_M1 |   | GPIO4_B2 | 27 | 28 | GPIO4_B3 |   |   | I2C4_SCL_M0 | I2S2_SDO_M1 | 139
107 |   |   |   |   | GPIO3_B3 | 29 | 30 | GND |   |   |   |   |  
108 |   |   |   |   | GPIO3_B4 | 31 | 32 | GPIO3_C2 | UART5_TX_M1 |   |   | I2S1_SDO3_M2 | 114
115 | UART5_RX_M1 |   | I2S1_SCLK_RX_M2 |   | GPIO3_C3 | 33 | 34 | GND |   |   |   |   |  
100 |   |   | I2S3_LRCK_M0 |   | GPIO3_A4 | 35 | 36 | GPIO3_A7 |   |   |   |   | 103
36 |   |   | I2S1_SCLK_RX_M0 |   | GPIO1_A4 | 37 | 38 | GPIO3_A6 |   |   |   | I2S3_SDI_M0 | 102
  |   |   |   |   | GND | 39 | 40 | GPIO3_A5 |   |   |   | I2S3_SDO_M0 | 101

influential-eliot, 2024-09-11:
Aha ... OK, thanks ... I'd not realised that they all use another connector other than the GPIO 😅  ... thanks, @Gadgetoid !

closed 2024-09-11 by influential-eliot

Gadgetoid, 2024-09-11:
DPI is via a GPIO alt mode, but it's probably not available on the RADXA's GPIO. AIUI the Pi is a bit of an outlier in that respect.

influential-eliot, 2024-09-11:
Gotcha, thanks ... also ... for the older Radxa models it would seem that it is definitely true:
https://forum.radxa.com/t/using-dpi-displays-on-the-radxa-zero/10108/2

I wonder if it's true for the new X4 which has a GPIO powered by the RPI RP2040 ... that could be a fun thing. I'll look into it. Thanks again!