pimoroni/hyperpixel4 #229: Raspbery Pi 5: HyperPixel 4.0 - black screen

dip39, 2025-02-20:
Hi,
I’m using HyperPixel 4.0, 4" TFT LCD display, 800x480, GPIO with Raspberry Pi 5.
I have installed it successfully and added to /boot/firmware/config.txt
`dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=90`

It was working perfectly: displaying text console... 
but after one day the screen stopped functioning.
When it boots, the backlight is on, but then (probably after booting the core) the backlight goes off and the screen remains black. When I run shutdown now the backlight again turns on, but still nothing on the screen.

Please help me to resolve the issue.

Here is the complete config.txt:

```
dtparam=i2c_arm=on
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1
[cm4]
otg_mode=1
[cm5]
dtoverlay=dwc2,dr_mode=host
[all]
dtparam=pciex1
dtparam=pciex1_gen=3
dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=90
```

```
~ $ uname -a
Linux naspi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```

```
~ $ dmesg | grep Goodix
[    7.533083] Goodix-TS 14-0014: supply AVDD28 not found, using dummy regulator
[    7.533139] Goodix-TS 14-0014: supply VDDIO not found, using dummy regulator
[    7.534176] Goodix-TS 14-0014: Error reading 1 bytes from 0x8140: -6
[    7.562627] Goodix-TS 14-0014: Error reading 1 bytes from 0x8140: -6
[    7.589592] Goodix-TS 14-0014: I2C communication failure: -6
[    7.589723] Goodix-TS 14-005d: supply AVDD28 not found, using dummy regulator
[    7.589754] Goodix-TS 14-005d: supply VDDIO not found, using dummy regulator
[    7.592858] Goodix-TS 14-005d: ID 911, version: 1060
[    7.631055] input: 14-005d Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-14/14-005d/input/input5
```

```
~ $ ls /dev/i2c-*
/dev/i2c-1  /dev/i2c-13  /dev/i2c-14  /dev/i2c-15
```

```
~ $ dmesg | grep vc4
[    3.836258] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    3.847342] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.057599] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.062120] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.066702] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.073348] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.078755] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.088305] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.114047] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.123974] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.128886] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    7.135628] rc rc0: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0
[    7.135677] input: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0/input1
[    7.179793] input: vc4-hdmi-0 HDMI Jack as /devices/platform/soc/107c701400.hdmi/sound/card0/input2
[    7.180775] vc4-drm axi:gpu: bound 107c701400.hdmi (ops vc4_hdmi_ops [vc4])
[    7.192734] rc rc1: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1
[    7.192788] input: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1/input3
[    7.212583] input: vc4-hdmi-1 HDMI Jack as /devices/platform/soc/107c706400.hdmi/sound/card1/input4
[    7.215783] vc4-drm axi:gpu: bound 107c706400.hdmi (ops vc4_hdmi_ops [vc4])
[    7.229836] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    7.229998] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    7.236481] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.237667] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.245831] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 1
[    7.248552] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    7.252075] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    7.255662] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
```

charel, 2025-03-05:
I have the same issue whenever I use Xorg. The issue is present with Ubuntu/DragonOS, Kali, and Parrot. 
Sometimes it helps to switch to a different tty (press CTRL+ALT+F2 or any other F key except F7 or F1, depending on which tty your greeter is starting). 

Once in a different tty you can install and switch to a wayland greeter and DE. 
For example, you can try installing gnome and use gdm3 as the greeter. Make sure that wayland is not disabled in your `/etc/gdm3/custom.conf` and after a successful boot to gdm3, select Gnome (not Xorg) for your session.

dip39, 2025-03-05:
> I have the same issue whenever I use Xorg. The issue is present with Ubuntu/DragonOS, Kali, and Parrot. Sometimes it helps to switch to a different tty (press CTRL+ALT+F2 or any other F key except F7 or F1, depending on which tty your greeter is starting).
> 
it didn't help at all. As I installed clean NON-GUI (no X server) I cannot even see the boot log and console. But previously I was able to see it...

closed 2026-08-25 by Gadgetoid